TinyTapeout/tt-demo-pcb #13: removing carrier board with the top header soldered is hard

mattvenn, 2024-07-15:
can't get my levers in there when the top header is soldered (the one next to usb)
if the carrier board was moved to be more central in the demo board, it would be easier to remove it by using levers

mattvenn, 2024-12-11:
with the latest configuration of components it's even harder to get the levers in

[image]

psychogenic, 2024-12-11:
[image]
The arrows are meant to indicate where to stick in your level sticks.

mattvenn, 2024-12-11:
I'm trying to say that it needs levering from the top side as well,
otherwise the pins still get bend

psychogenic, 2024-12-11:
I have to 3D print some of your thingies, I'd thought you could inch along but they must be too short.

mattvenn, 2024-12-11:
I can get them under, but then they normally are pushing against a component instead of the board. Maybe this is actually fine, I'm just worried about popping off a component or cracking a cap

psychogenic, 2024-12-11:
It's not ideal.  Best thing might be to make the carriers a bit wider so you have a lip to use from the outside.

closed 2025-02-27 by mattvenn