TinyTapeout/tt-demo-pcb #22: Components interfere with levering out the breakout board: make space somehow

psychogenic, 2025-02-01:
U2, the flash, and the lack of clarity on where to stick the 3-printed levers when removing the breakout.  Try and make some space and indicate the path better than the small arrows in use now:

[image]

mattvenn, 2025-02-01:
Have you tried the levers? They work much better at the top and bottom of
the breakout, not both at the bottom (where they cause pin bendage at the
top)

MichaelBell, 2025-02-01:
As Matt notes, the bottom side is no problem.  A similar clear path on the top side is also needed to lift the carrier off evenly without risking pin bendage.

With the current design you can just about get one tool in behind the USB port on the top side, but it's a bit fiddly and seems non-zero risk to the port and the diodes beneath the port.

psychogenic, 2025-02-11:
Wanted to keep the bigger regulator and not disrupt things too badly, wound up with this.  
I finally printed out the levers and they don't need to go too deep to function, we basically just need access to the four corners, which this should allow without too much issue
[image]

mattvenn, 2025-02-12:
apart from adding the arrow, it looks no different to my TT06 board - what changed?
for me when I try to put in the levers at the top, on the right the usb socket gets in the way, and on the left the regulator gets in the way

psychogenic, 2025-02-12:
> apart from adding the arrow, it looks no different to my TT06 board

Well I never!1!

It's the most I could do without moving a ton of wiring, but still got a couple of mm clearance--heat map of changes below.

The fact is, though, I can send a video of how I've been doing it with current boards: if you come in "on the side" and twist it works, and then you only have an issue when your lever is down touching the PCB, which this I think is enough to change.

Also, with 60% infill we could prob make the things thinner without issue.

[image]

mattvenn, 2025-02-12:
good point on the width, they probably can be thinner

closed 2025-02-27 by mattvenn